VHDL-LS/rust_hdl_vscode #2: Document procedure for building, installing and running the extension.

Bochlin, 2018-12-02:


Xcodo, 2019-03-22:
I'm interested in giving this a try and can write up some documentation while I'm at it.
Can you give me a rough idea on how to get started? (I already tried cloning the extension by got errors about the `client/out/` folder).

Bochlin, 2019-03-23:
Compile the language server according to [https://github.com/kraigher/rust_hdl]() and place the vhdl_ls binary in a directory named "server" at the project root.

Install [Node.js](https://nodejs.org/en/)

In the project root run
> npm install
> npm run compile

In VsCode you can now start the build task (ctr-shift-b).
Press F5 to open the debug window and then run Launch Client.

I have note really had any tme took look at this lately and there is nothing in place for publishing the extension.

Xcodo, 2019-03-26:
👍 I got this running today. I'll do a pull request for an updated README then look into publishing

GlenNicholls, 2019-10-18:
What is the status of this, @Xcodo ?

GlenNicholls, 2019-10-19:
@Bochlin when you say put the vhdl_ls in the project root, the root of vs code or the VHDL project?

Bochlin, 2019-10-19:
Relative the root of the plugin project: 
<plugin_project>/server/vhdl_ls

There is an option in the plugin to use the docker image provided by rust_hdl (on linux) and I think that it should also work by compiling with cargo and then the plugin will run it from cargo's install directory. You can change the location of the vhdl_ls binary in the plugin settings under extensions/VHDL.

Bochlin, 2019-12-03:
Added CONTRIBUTING.md with description for project setup

closed 2019-12-03 by Bochlin