VHDL-LS/rust_hdl_vscode #11: Case syntax should insert all case options depending on the enumerate defined for the case select variable

GlenNicholls, 2020-01-15:
When using the `case` syntax, a nice feature would be to automatically insert all case names. To accomplish this, the parser would need to see that the `case` variable as shown below is an enumerate, and if so, it would automatically insert all options:

```
type fsm_t is (IDLE, STEP_1, DONE);
signal state : fsm_t := IDLE;

...

process(all)
begin
    case state is -- after user presses tab when entering case select variable, it will auto populate options like below only if parser recognizes signals type as an enumerate
        when IDLE =>

        when STEP_1 =>

        when DONE =>

        when others => null; -- could have option for the user to set state to first option in enumerate for the others case
    end case;
end process;

Bochlin, 2020-01-15:
Good feature.  
This type of completion must be implemented in the language server as it would require the client to more or less implement a full parser and library management.

GlenNicholls, 2020-01-20:
@Bochlin Do you want me to leave this open here or move it to rust_hdl?

Bochlin, 2020-01-26:
@GlenNicholls You can move it to rust_hdl since none of the functionality would be in this extension.

GlenNicholls, 2020-01-28:
Opened issue here https://github.com/kraigher/rust_hdl/issues/70

closed 2020-01-28 by GlenNicholls