VHDL-LS/rust_hdl_vscode #14: Startup of vhdl_ls client fails silently when vhdl_ls.toml file is missing.

pvanschendel, 2020-03-05:
Issue seems to be caused by commit `a288ebf0b74d928aef3c61d00e63876e62bc2d3e`.

The startup of the client fails when *vhld_ls.toml* does not exist in the current workspace root (extension.ts:110).

I am not sure if the failure is a bug: 
The vhdl_ls server will also output an error message if the failure is fixed, but I think startup of the client should at least not fail silently.

GlenNicholls, 2020-03-05:
IMO, I don't think this extension should require a `vhdl_ls.toml` in the first place, or it should at least spit out an error but still be able to parse single files. On some of the projects I'm working on right now, I don't need the full capability of the extension, just syntax highlighting on single files as I'm working on numerous projects and it is tedious to set each one up to have a `vhdl_ls.toml`. Because I'm splitting my time between so many projects, it's not practical to have multiple VSC windows open in the project workspace, so I'd rather just live for the time being with single-file analysis

Bochlin, 2020-03-05:
I get a silent failure when opening a file directly from the filesystem. The intention is that it should load the server and run without a library configuration. I can however open a folder without a vhdl_ls.toml without the client failing (the server will still output an error that vhdl_ls.toml) is missing.  

The extension should work without the configuration.

Bochlin, 2020-04-05:
Client was crashing when attempting to setup a file watcher for the vhdl_ls.toml file relative to the workspace folder. This should be solved in v0.3.1.

closed 2020-04-05 by Bochlin