VHDL-LS/rust_hdl_vscode #15: Alternative configuration with VUnit run.py instead of vhdl_ls.toml

Gzah92, 2020-03-10:
It would be great if it was possible to use a VUnit run.py file as configuration source instead of the vhdl_ls.toml. The common VUnit distributions contain commonly used libraries, and all project source code is associated to libraries in the file as well. In the current way absolute paths have to be used to specify the location of e.g. the location of OSVVM, making the file not portable and not a part of any repository, increasing setup time.
While it should be really simple to adapt the run.py to create the vhdl_ls.toml, maybe this could as well be directly integrated into the extension?
Would also fit nicely with the vunit-test-explorer.
Thanks for your consideration!

Bochlin, 2020-04-01:
Sounds like a decent enhancement. I think that this should be an option which is disabled by default and if enabled, regenerate the vhdl_ls.toml file each time the run.py is executed in order to update the test cases.

jakobjung10, 2024-01-21:
@Gzah92 Creating the vhdl_ls.toml from a run.py is supported in [VHDLbyHGB](https://marketplace.visualstudio.com/items?itemName=P2L2.vhdl-by-hgb&ssr=false#overview)