VHDL-LS/rust_hdl_vscode #16: Please allow suppression for messages related to vhdl_ls.toml

nversluis, 2020-03-27:
I would like to be able to use this extension without specifying a vhdl_ls.toml file.
In fact, the extension works very well without it being able to load the aforementioned file already.

However, every time I open VS Code, the following messages pop up:

> Loaded Installation configuration file: /home/<username>/.vscode-server/extensions/hbohlin.vhdl-ls-0.3.0/server/vhdl_ls/0.15.0/vhdl_ls-x86_64-unknown-linux-gnu/bin/../vhdl_libraries/vhdl_ls.toml
> [Error - 11:56:09 AM] Found no vhdl_ls.toml config file in the workspace root path: No such file or directory (os error 2)
> [Warn  - 11:56:09 AM] Found no library mapping, semantic analysis disabled, will perform syntax checking only
> [Info  - 11:56:09 AM] Opening file /home/<user>/path_to_file/file.vhd that is not part of the project

I would like to be able to suppress these messages, so they don't pop up every time I launch VS Code, since they don't impact what I am using the extension for, which is using it on a per-file basis.

Bochlin, 2020-04-01:
It is the [language server](https://github.com/kraigher/rust_hdl) which sends these messages. As far as I know there is no way for the client to suppress messages from the server.

nversluis, 2020-04-01:
Would it then be possible to add an option to disable the language server output to VS Code entirely?
Honestly the only thing that bothers me is that the extension pops up the VS Code terminal every time I start it.

levi-ee, 2022-05-18:
It's a good VS Code extension, but I totally agree with @nversluis.
Did someone found a easy work around for that?