VHDL-LS/rust_hdl_vscode #18: vhdl_ls.toml not working with soft link in workspace root directory path

exe1099, 2020-05-09:
Create a directory containing top.vhd and vhdl_ls.toml with
```
[libraries]
work.files = [
'./*.vhd',
]
```
Create a soft link to that directory, open it in vscode. Opening top.vhdl results in
```
[Info  - 7:57:12 PM] Opening file .../Linked/top.vhd that is not part of the projec
```
while opening the file via the original directory doesn't throw any error.

Bochlin, 2020-05-22:
This is a function of the language server. Would you mind posting it at https://github.com/kraigher/rust_hdl/issues?  

This issue can be kept open until resolved in the language server.

kraigher, 2023-11-13:
This could potentially be fixed now by https://github.com/VHDL-LS/rust_hdl/issues/115. Try the latest release.

closed 2023-11-13 by kraigher