VHDL-LS/rust_hdl_vscode #20: vhdl_ls.toml not found on VSCode portable configuration

Mathinh0, 2020-06-24:
I would like to create a vhld_ls.toml file for my configuration, but it is never found by VHDL LS. 

I'm using VSCode in its Windows portable version, without any .vscode folder. The user data are included in data folder in the main folder of VSCode.
I'm trying to put vhdl_ls.toml config file in several folders (./data, ./data/user-data, ./data/user-data/User), but it's never loaded by VHDL LS. The installation config file is always loaded (file given with VHDL LS extension).

I'm trying too to put vhdl_ls.toml config file in my workspace folder (with the .code-workspace file), but same result as other tests.

What can I try to succeed my VHDL LS configuration ?

Thx

am9417, 2020-06-24:
I discovered this extension at the same time as you did, probably. I had the same issue because output complained about missing vhdl_ls.toml file.

I got it working by just opening a single folder (no "new workspace", no `.vscode` directory) in VS code and the `vhdl_ls.toml` file then placed in the root of that directory. Changing the toml-file required restarting the VHDL LS extension (command palette: *VHDL LS: Restart VHDL LS*).

Mathinh0, 2020-06-25:
Great !! It's working.
Thx for your support am9417.

I'm also interesting to know where I can place the `vhdl_ls.toml` file in user data folder with VSCode portable version.

Another question : Is it possible to define several `vhdl_ls.toml` files in a same project ? For example, one to define the general pkg (std, ieee,...) and another one to define the specific pkg of a project.

Thx

ericjmcd, 2020-06-25:
I had a similar issue but with a multi-folder workspace.  It turned out that the workspace "root" was actually the root of the first folder in the workspace (if you opened the vscode workspace file you could see/edit the order).  After changing that and restarting VSCode it worked for me.

I was using VSCode on Windows with RemoteSSH into Ubuntu18.04.

Bochlin, 2020-06-30:
The language server will load configuration files in the following order of priority (first to last):
1. A file named .vhdl_ls.toml in the user home folder.
2. A file name from the VHDL_LS_CONFIG environment variable.
3. A file named vhdl_ls.toml in the workspace root.

There is no automatic update currently for configrations changes. The clien will send a notification to the server, but the server does not implement any response to it. It would be possible to restart the server for changes in the configuration but since restarting the server prompts a full reparsing of the project I have been reluctant to do it that way.

@ericjmcd The extension does not support multi-root workspaces as of yet.

closed 2022-03-30 by Mathinh0