VHDL-LS/rust_hdl_vscode #21: No action on hover.

m-kru, 2020-06-30:
There is not action on hover, no pop with additional information.
Is this because of lack of support in [rust_hdl](https://github.com/kraigher/rust_hdl), or because lack of support in this repository?

Bochlin, 2020-06-30:
Lack of support in rust_hdl.
I have actually made a local test for a hover implementation for the server but have not made a pull request yet. I have an open pull request for outline which I would like to finish first. In that PR there are functions which can be used for the hover to extract snippets from the code.

m-kru, 2020-06-30:
What is `semi_token` representing?

Bochlin, 2020-06-30:
The semi colon at the end of the statement or declaration. If you have a deeper look into the code, you will find that there will be comments attached to this token, which could be used for "documentation" comments in the hover.

kraigher, 2021-03-07:
Hover is now supported in the language server

closed 2021-03-07 by m-kru