VHDL-LS/rust_hdl_vscode #23: Problems with user language server location.

m-kru, 2020-07-05:
I have cloned rust_hdl and built it locally. In the Vhdlls settings I have user language server location. I have provided correct path. Based on the logs from vhdl_ls I can see, that it receives requests and sends responses:
```bash
05 Jul 12:07:27:783 [0m[37m[TRACE] [0m(2) vhdl_ls::stdio_server: [vhdl_ls/src/stdio_server.rs:147] GOT REQUEST: "{\"jsonrpc\":\"2.0\",\"id\":0,\"method\":\"initialize\",\"params\":{\"processId\":28233,\"rootPath\":\"/home/mkru/workspace/smx\",\"rootUri\":\"file:///home/mkru/workspace/smx\",\"capabilities\":{\"workspace\":{\"applyEdit\":true,\"workspaceEdit\":{\"documentChanges\":true,\"resourceOperations\":[\"create\",\"rename\",\"delete\"],\"failureHandling\":\"textOnlyTransactional\"},\"didChangeConfiguration\":{\"dynamicRegistration\":true},\"didChangeWatchedFiles\":{\"dynamicRegistration\":true},\"symbol\":{\"dynamicRegistration\":true,\"symbolKind\":{\"valueSet\":[1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,23,24,25,26]}},\"executeCommand\":{\"dynamicRegistration\":true},\"configuration\":true,\"workspaceFolders\":true},\"textDocument\":{\"publishDiagnostics\":{\"relatedInformation\":true},\"synchronization\":{\"dynamicRegistration\":true,\"willSave\":true,\"willSaveWaitUntil\":true,\"didSave\":true},\"completion\":{\"dynamicRegistration\":true,\"contextSupport\":true,\"completionItem\":{\"snippetSupport\":true,\"commitCharactersSupport\":true,\"documentationFormat\":[\"markdown\",\"plaintext\"],\"deprecatedSupport\":true,\"preselectSupport\":true},\"completionItemKind\":{\"valueSet\":[1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,23,24,25]}},\"hover\":{\"dynamicRegistration\":true,\"contentFormat\":[\"markdown\",\"plaintext\"]},\"signatureHelp\":{\"dynamicRegistration\":true,\"signatureInformation\":{\"documentationFormat\":[\"markdown\",\"plaintext\"],\"parameterInformation\":{\"labelOffsetSupport\":true}}},\"definition\":{\"dynamicRegistration\":true,\"linkSupport\":true},\"references\":{\"dynamicRegistration\":true},\"documentHighlight\":{\"dynamicRegistration\":true},\"documentSymbol\":{\"dynamicRegistration\":true,\"symbolKind\":{\"valueSet\":[1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,23,24,25,26]},\"hierarchicalDocumentSymbolSupport\":true},\"codeAction\":{\"dynamicRegistration\":true,\"codeActionLiteralSupport\":{\"codeActionKind\":{\"valueSet\":[\"\",\"quickfix\",\"refactor\",\"refactor.extract\",\"refactor.inline\",\"refactor.rewrite\",\"source\",\"source.organizeImports\"]}}},\"codeLens\":{\"dynamicRegistration\":true},\"formatting\":{\"dynamicRegistration\":true},\"rangeFormatting\":{\"dynamicRegistration\":true},\"onTypeFormatting\":{\"dynamicRegistration\":true},\"rename\":{\"dynamicRegistration\":true,\"prepareSupport\":true},\"documentLink\":{\"dynamicRegistration\":true},\"typeDefinition\":{\"dynamicRegistration\":true,\"linkSupport\":true},\"implementation\":{\"dynamicRegistration\":true,\"linkSupport\":true},\"colorProvider\":{\"dynamicRegistration\":true},\"foldingRange\":{\"dynamicRegistration\":true,\"rangeLimit\":5000,\"lineFoldingOnly\":true},\"declaration\":{\"dynamicRegistration\":true,\"linkSupport\":true}}},\"trace\":\"off\",\"workspaceFolders\":[{\"uri\":\"file:///home/mkru/workspace/smx\",\"name\":\"smx\"}]}}"
05 Jul 12:07:27:801 [0m[37m[TRACE] [0m(3) vhdl_ls::stdio_server: [vhdl_ls/src/stdio_server.rs:152] SEND RESPONSE: "{\"jsonrpc\":\"2.0\",\"method\":\"window/logMessage\",\"params\":{\"message\":\"Loaded Installation configuration file: /home/mkru/workspace/rust_hdl/target/debug/../../vhdl_libraries/vhdl_ls.toml\",\"type\":4}}"
05 Jul 12:07:27:801 [0m[37m[TRACE] [0m(3) vhdl_ls::stdio_server: [vhdl_ls/src/stdio_server.rs:152] SEND RESPONSE: "{\"jsonrpc\":\"2.0\",\"method\":\"window/logMessage\",\"params\":{\"message\":\"Loaded workspace root configuration file: /home/mkru/workspace/smx/vhdl_ls.toml\",\"type\":4}}"
05 Jul 12:07:28:763 [0m[37m[TRACE] [0m(3) vhdl_ls::stdio_server: [vhdl_ls/src/stdio_server.rs:152] SEND RESPONSE: "{\"jsonrpc\":\"2.0\",\"result\":{\"capabilities\":{\"declarationProvider\":true,\"definitionProvider\":true,\"referencesProvider\":true,\"textDocumentSync\":2}},\"id\":0}"
```
However, if I right click on the entity name, type name etc. in the VSCode the `Go to Definition`, `Go to Declaration`, `Go to References`  etc. are gone. Do you know what might be wrong? @kraigher maybe you know what is going on?

kraigher, 2020-07-05:
You must have a library mapping for goto definition etc to work. Without library mapping only syntax is checked. Map your files using a vhdl_ls.toml file.

m-kru, 2020-07-05:
I have vhdl_ls.toml, with embedded vhdl_ls everything works. The problem arises when I try to use user language server location.

kraigher, 2020-07-05:
See this build flag: https://github.com/kraigher/rust_hdl/pull/51

m-kru, 2020-07-06:
This is something different. I am building with `packaged` feature. The `Go to Definition` etc. are not even present after right click. Also the syntax is not checked. Like the server is not loaded, although the log suggests it is loaded. The log comes from the server, maybe VSCode somehow does not receive the responses, when user language server location is used?

m-kru, 2020-07-06:
Ok, it is somehow related to the `simplelog` usage. I have replaced env_loger with simplelog. When I comment out simple log initialization from main.rs:
```Rust
    TermLogger::init(
        LevelFilter::Trace,
        ConfigBuilder::new()
            .set_time_format_str("%d %b %T:%3f")
            .set_level_padding(LevelPadding::Left)
            .to_owned()
            .build(),
            TerminalMode::Stdout
    )
    .unwrap();
```
and build the `vhdl_ls` it works. When I uncomment above initialization code, it stops working.
@kraigher any ideas what might be wrong?

kraigher, 2023-01-03:
The problem is you are logging to stdout in a language server that uses stdout to communicate to the client with json rpc.

closed 2023-01-03 by kraigher