VHDL-LS/rust_hdl_vscode #31: Language server upgrade failed

Xcodo, 2020-10-03:
There was a new version of the `vhdl_ls` released today. The auto update in VSCode failed:

> Checking for language server executable in c:\Users\mark_\.vscode\extensions\hbohlin.vhdl-ls-0.3.1\server\vhdl_ls
> Found version 0.15.0
> Using embedded language server
> Checking for updates...
> Language server started
> Current vhdl_ls version: 0.15.0
> Latest vhdl_ls version: 0.16.0
> Fetching https://github.com/kraigher/rust_hdl/releases/download/v0.16.0/vhdl_ls-x86_64-pc-windows-msvc.zip
> TypeError: Expected signal to be an instanceof AbortSignal
> Error: Language server download timed out after 60000.00 seconds.
> Language server update finished.

I can download the zip file at that URL fine.

My setup is
- VHDL LS extension v0.3.1
- VSCode v1.49.3
- Windows 10

I might have a go at debugging it tonight unless you say you're on it @Bochlin.

Bochlin, 2020-10-03:
I think that the problem is line 329 and 334 in `extension.ts`.   
The `abortController` is what should abort the download in case of a timeout, but I think that there is a type error at line 334 causing a `TypeError` exception (the `signal` parameter gets an argument of the wrong type). Not sure of what has changed since I wrote code that broke it but it is probably something in the dependencies,

I might be able to find some time tonight to investigate but if you have some time please feel free to see if you can find a solution.

```
329        const abortController = new AbortController();
        const timeout = setTimeout(() => {
            abortController.abort();
        }, timeoutMs);
        let download = await fetch(browser_download_url, {
334            signal: abortController.signal,
        }).catch((err) => {
            output.appendLine(err);
            throw new Error(
                `Language server download timed out after ${timeoutMs.toFixed(
                    2
                )} seconds.`
            );
        });
```

Bochlin, 2020-10-04:
Problem is the webpack packaging which mangles the class name of the AbortController.signal property in combination with a type check in node-fetch. I'll try to solve this today.

Bochlin, 2020-10-04:
Should be resolved in v0.3.2, which also include the 0.16.0 server.

Xcodo, 2021-03-05:
Finally got an update that I could verify this! Works like a charm now, thanks @Bochlin.

closed 2021-03-05 by Xcodo