VHDL-LS/rust_hdl_vscode #36: VHDL LS github organization

kraigher, 2021-03-20:
I made a new VHDL-LS github organization https://github.com/VHDL-LS/rust_hdl. My hope is that you would consider moving this repository to the organization. I send you an invite to the organization.

PS: I also designed a new logo.

Bochlin, 2021-03-21:
Makes sense to me to have them under the same umbrella so I can make the move.

One thing I would like to consider in that case is to change the publisher of the extension on the VSCode marketplace to a more generic "VHDL-LS" (or similar) publisher maintained by the VHDL-LS org. There are a couple of ways of doing that but the simplest seems to be to simply republish the extension under the the publisher, see https://github.com/Microsoft/vscode/issues/3670.  It seems to be possible to have several users attached to the same publisher with different user rights.

Changing the publisher is not pre-condition for moving the repo.

kraigher, 2021-03-21:
Sound good.

Bochlin, 2021-03-21:
Forgot to mention, the logo looks good and I will happily replace the existing one.

kraigher, 2021-03-23:
Great, the logo is part of the repo as an svg. Let me know if you need another format.

Bochlin, 2021-07-18:
Better late than never, but the repository is transferred now...

I would need an 128x128 PNG icon in order to update as VSCode does not render SVGs due to security concerns.

closed 2023-01-03 by kraigher