VHDL-LS/rust_hdl_vscode #45: go to definition not working

Ahmad-Zaklouta, 2021-11-17:
when vscode does not find vhdl_ls.toml file, go to definition work. But when there is vhdl_ls.toml file go to definition is not working.
Is there a configuration that I should do for that?

GlenNicholls, 2021-11-19:
I think your issue is that you’re putting files in library work… work is not a library that you should try and add files to. When you reference work in VHDL:

```
Library lib;

…

    Inst: entity work.mult
```

work is referencing the current library. Work itself is not a library though. Please try changing the library to something else.

Work is kind of similar to the STD library. It’s reserved by VHDL and is implied. By this I mean that you do not need to add a library declaration to reference them nor do you add files to them; a simple `use work.some_pkg;` will suffice.

You can find free copies of the LRM online, I definitely recommend reading through it to understand this better. I’m on mobile so I only provided the bare minimum because typing is tedious.

Ahmad-Zaklouta, 2021-11-20:
Thanks for your answer.
I worked now. I had an entity that has two architecture and they had the same name. so it didn't work.
another thing, what do they mean by the workspace root? is it the project root directory or where I save my vscode workspace file?

GlenNicholls, 2021-11-24:
Workspace root is considered wherever VSCode is opened. Essentially File>Open Folder I believe is the pull-down (on mobile so can’t double check that.) Not sure if you’ve seen this yet, but the README for rust_hdl also provides information about the order in which it searches for the TOML config file for the project

closed 2023-01-03 by kraigher