VHDL-LS/rust_hdl_vscode #46: Valid VHDL marked as error

deanhumphreys, 2021-11-23:
Hi,

The attached VHDL (generate else) is marked as error, however it is valid and accepted by both Questasim and Quartus.
I am working on a database that uses this format in many places, so it will be difficult to change it everywhere.
Would you be able to fix this?

[image]

Thanks,
Dean

deanhumphreys, 2021-11-23:
Looks like this is a similar issue:

[image]

GlenNicholls, 2021-11-24:
I'm not sure why your second example shows an error, but for the first one it looks really bizarre to me with that stray `end;`. That might be valid, but I'm suspicious that it will lead to weird errors. Reference https://www.doulos.com/knowhow/vhdl/vhdl-2008-easier-to-use/#generate

I haven't dug through the LRM for that specifically, but I would personally side with rust_hdl on that one. Also, you may need to go through other channels (I.e. TerosHDL https://gitter.im/TerosHDL/community) to get traction on this. I don't see very much movement on this extension or the underlying tool rust_hdl.

With all of that said, Kraigher mentioned that there is VHDL 2008 capability in rust_hdl. However, It's entirely possible that corner cases haven't been fully tested before he ended up changing career paths and not focusing on this. I've noticed that certain keywords might throw the highlighting off which I believe was caused by this extension and not the LS, but It's been a while since I inquired about that.

deanhumphreys, 2021-11-24:
Fair enough.
So what you're saying that it's unlikely this will be adressed?
Is there a way to waiver error types? Since this error marks the whole file as errored and stops parsing the file, it's therefore un-referenceable.

Any workaround you can suggest, aside from changing the codebase?
/Dean

GlenNicholls, 2021-11-24:
Not necessarily, I would suggest trying TerosHDL (which can use GHDL or rust_hdl as the VHDL LS). Your TOML file will still work, and TerosHDL has much much more capability. That project is active and they would be more reachable to provide a fix if the issue is present there. Your next best option if you want a solid VHDL extension for vscode would then be to use Sigasi's extension. Last I heard it is still in Beta and it also costs money after the trial ends.

No, it does not support reducing the severity of errors to my knowledge.

I have used this extension and there is only one case I can think of that is a legitimate bug. I would just recommend fixing your code because that stray `end;` in your `if/else` generate is confusing and will make it harder to read when they get more complicated. Reference https://www.edaplayground.com/x/389T for a proper example. The way I see it is that yes that code may be present in a lot of places and valid because `begin` may be optional (not positive on that), but it is bad style because you aren't providing the generate with a declaration region using `begin`. If I saw that in a code review, I wouldn't approve the pull request because `begin` and `end` in VHDL are brackets and should be used as such. With c/c++ for example, even if `{` was optional, that would absolutely not be recommended to treat as optional if you provided the closing brace `}`. 

As for the `case` generate, I am generally against using an enumerate (`MODE` in your case) for generics... IMO, the code you provided looks like you have two generics (one for MODE and another for pipeline depth) which is redundant and confusing. A shift register of depth zero is known to be pass-through, depth of one is obvious that it's a single pipeline stage, and so on. Enumerate's make sense in some situations like FSM's so you can provide intuitive names, but this just feels needlessly complicated for something easily accomplished by a simple integer generic+`if/elsif/else` generate.
```
entity shift_reg is
  generic (
    PIPELINE_STAGES : integer range 0 to 5 => 0  
  )
...
```

rust_hdl (LS this extension uses) does support VHDL 2008 for the most part (maybe all, not 100% sure), however, this extension to my knowledge doesn't support all 2008 patterns. The `case` generate likely is confusing the extension because it doesn't expect a case statement outside a process. The fact it didn't mark an error earlier leads me to believe it is fine with the case outside a process but is not expecting a declaration region within each case. This is likely the same reason the `if/generate` is also giving an error.

Lastly, I don't use rust_hdl as a golden reference for errors because it cannot handle elaboration. I wouldn't say it is not "referencable". The power I find is that I don't have to wait to run a testbench/build before I get feedback about analysis errors and I can navigate large code bases much easier with go-to definition. Yes, there are minor issues like this that are inconvenient, but because the maintainers aren't super active, it would be up to the community to open PR's and add new tests. I'm sure they would be more than happy to mentor or review if people are interested in contributing.

GlenNicholls, 2021-11-24:
A simple fix for the first issue might be to just add the `begin` so that the LS sees the matching "bracket":
```
gen_ram: if AW > 0 generate
begin
...
end; -- if allowed, I would remove this end; entirely because it feels redundant
else generate
...
end generate gen_ram;
```

If you don't have the begin, then there shouldn't be an end in the first place if that makes sense. The lack of `end` if you have a begin makes much more sense because you next bracket starts with `else/elsif` and the final end is already provided. It's kind of awkward, so I'm wondering if there's something more to those decisions. Might be worth providing your example about this at https://gitter.im/vhdl/General or https://gitlab.com/IEEE-SA/vasg