VHDL-LS/rust_hdl_vscode #55: Allow specifying vhdl_ls version to use

wrightsg, 2022-12-21:
When selecting `embedded` for the language server, the plugin will always use the latest `vhdl_ls` release.

This can break stuff without actually changing anything. In my case, I'm running the VHDL LS plugin on Ubuntu 18.04. The latest `vhdl_ls` release `0.22` which now required `glibc` version `> 2.27`. Ubuntu 18.04 only comes with `glibc` version `2.27` so I cannot use the VHDL LS plugin anymore.

It would be nice to be able to specify an explicit `vhdl_ls` version with the `embedded` option.

kraigher, 2022-12-29:
It would probably be better if we build `vhdl_ls` using a bit older docker image so that it does not require such a new glibc. Currently it is build using `ubuntu-lastest`

kraigher, 2022-12-29:
I did some investigation and github is about to deprecate the ubuntu-18:04 runner so it does not seem attractive to use it. You could always build vhdl_ls yourself. I believe the vscode plugin supports pointing out binaries you build yourself. 

Using an old version of vhdl_ls is not good since a lot of new functionality is being added right now.

Bochlin, 2022-12-30:
Yes you can use your own compiled language server, just point it out with the `vhdlls.languageServerUserPath` option.

kraigher, 2022-12-30:
Another option is to build the binaries without any libc dependency using the musl target. That would reach the most possible systems. If someone makes a PR of that it probably would be approved.

kraigher, 2023-01-03:
Yes cargo install just adds the binary. Cargo has no concept of a data folder. When building locally you should just check out the code and run cargo build --release and point to the binary in the target/release folder and it will find the libraries.

Anyway we should just build a musl version that does not depend on glib and avoid this complexity for users such as you.

kraigher, 2023-01-03:
Note that it is not recommended to mix vhdl_libraries folder with a vhdl_ls binary when they are not from the same commit. The standard.vhd package is tightly coupled with the binary and has recently changed as the vhdl_ls analysis became smarter.

kraigher, 2023-01-03:
@Bochlin I am adding `x86_64-unknown-linux-musl` to the github release targets with a plan to phase out `x86_64-unknown-linux-gnu`. Is it possible to make a new `rust_hdl_vscode` release that uses the `musl` binary?

kraigher, 2023-01-03:
@Bochlin https://github.com/VHDL-LS/rust_hdl/releases/tag/v0.24.0 now includes musl builds which do not have any glibc dependency on Linux

kraigher, 2023-01-03:
So rust_hdl_vscode should switch from linux-gnu to linux-musl zip folder on linux.

Bochlin, 2023-01-03:
Published 0.4.0 which uses the musl build instead.

kraigher, 2023-01-03:
@Bochlin excellent 

@wrightsg could you try if it solved your original problem?

wrightsg, 2023-01-09:
I can confirm that the extension works again on Ubuntu 18.04 with version `v0.4.0`.

Thank you!

closed 2023-01-09 by wrightsg