VHDL-LS/rust_hdl_vscode #61: GitHub Actions Release job does not work

kraigher, 2023-02-15:
It seems the release pipeline in Github Actions is not able to perform a release. I wanted to update the readme but could not publish the extension. @Bochlin Is this something you could help with?

I think it could be related to the release pipeline running node v10.

Bochlin, 2023-02-20:
Made the last release locally and manually and had to work around some Node issues, I should be able to make a fix for those. 

There is also another issue with publishing which is the need for an access token to my publisher on Azure. I can create a new token but I don't think that I have access to add secrets to the repo.

I think that a better option might be for you to create an organization for VHDL LS on Azure and that we republish the extension under that organization. That way you get proper ownership of the extension. You should also be able to invite me to the organization and I can support with the setup.

In any case, I can fix the release job for now, I just need a way to get the access token into the GitHub secrets.

kraigher, 2023-02-20:
I bumped your role from "write" to "maintain", it claims to give you access to repository settings.

kraigher, 2023-02-20:
We could move the extension but it will be a bit of a hassle for us and the users: https://stackoverflow.com/questions/69981814/transfer-ownership-of-a-vscode-extension. Basically it seems like you cannot move an extension but have to create a new one and deprecate the old.

I am fine with having it the way it is if I can independently make changes to the readme and make releases via Gitlab. When I implement autocomplete I will need to move the snippet definitions from the client to the server and it could result in a few releases to move the responsibility.

That is unless you have a reason to want it otherwise? I do not know of interested you are in maintaining this. For me it is a huge help that you do as I can focus my time on the server itself and not have to also make a client.

Bochlin, 2023-02-23:
No, you cannot move and have to republish, you can mark the old as deprecated and link to the new. One of the reasons is that I have another extension published on the same org and it would make it a little cleaner to separate. 

I don't mind maintaining it the way it is though and will fix the publishing as soon as I get a little time over in the next days.

kraigher, 2023-02-23:
I can setup a new organisation so that you can have a clean separation with the other extension. But it would be good to get the existing CI working first.

kraigher, 2023-03-08:
@Bochlin I was thinking it would be good to have the current CI working before we try creating a new extension under another name. Just saying so you are not waiting for me to do anything.

Bochlin, 2023-03-09:
Sorry, not waiting, just being incredibly slow. From what I can see the package job is working but the secret for publishing is still missing. I have created a token now but it seems that you need `admin` access to add it o(https://docs.github.com/en/actions/security-guides/encrypted-secrets#creating-encrypted-secrets-for-a-repository). Do you have a good way I could send it to you?

Bochlin, 2023-03-09:
Token is updated. In order to run the publish job you need to update `"version": "0.4.0"` to a new version in [package.json](https://github.com/VHDL-LS/rust_hdl_vscode/blob/master/package.json) which will update the version of the package. The job itself is triggered by tagging the repo with the same version (i.e. `v0.4.1`) on `master`.

kraigher, 2023-03-12:
Got the following error when running just now:
https://github.com/VHDL-LS/rust_hdl_vscode/actions/runs/4398509842/jobs/7702419030

```
Run vsce publish -p $PUBLISH_TOKEN --packagePath vhdl-ls/vhdl-ls-${GITHUB_REF:11}.vsix
/opt/hostedtoolcache/node/10.24.1/x[6](https://github.com/VHDL-LS/rust_hdl_vscode/actions/runs/4398509842/jobs/7702419030#step:6:7)4/lib/node_modules/vsce/out/package.js:136
    return (translations ?? [])
```

I saw that the release pipeline uses node version 10, isnt that quite old?

kraigher, 2023-03-12:
Works now after bumping the node version.

closed 2023-03-12 by kraigher