VHDL-LS/rust_hdl_vscode #64: Problems with executing the extension

jakobjung10, 2023-03-04:
Hello!

I am having problems with installing rust_hdl_vscode.
I cloned the repository, but I am facing errors when i execute the extension.
My first guess would be that i am missing some files, that are required for execution.

I know that the rust_hdl_vscode repository is mainly responsible for the VS Code extension and the actual language Server is on the rust_hdl repository.

I am also a little bit confused by the paths in the extension.ts file.

`const languageServerDir = ctx.asAbsolutePath(
        path.join('server', 'vhdl_ls')
    );`

`languageServer = path.join(
        'server',
        'vhdl_ls',
        languageServerVersion,
        languageServerName,
        'bin',
        languageServerBinaryName + (isWindows ? '.exe' : '')
    );`

In the above code samples from the extension.ts file, the path includes a folder named 'server', but i can't find a folder with this name in the extension. Is this something that should be automatically created or rather be manually created.

I am very new to the topic and would be grateful, if someone could explain how to get the extension running and files i am missing etc.
I am interested in the project, but haven't found any proper documentation yet.

kraigher, 2023-03-04:
The extension downloads the latest language server from github release artifacts. In that way the server is always up to date without having to republish the extension itself.

jakobjung10, 2023-03-04:
When i execute the extension, i get an error. "The preLaunchTask 'npm: webpack' terminated with exit code 1.".
By pressing "Debug anyway", i get the errors like below:

Activating extension 'hbohlin.vhdl-ls' failed: Cannot find module 'c:\Users\Jakob\Documents\GitHub\rust_hdl_vscode\dist\extension' Require stack: - c:\Users\Jakob\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\loader.js - c:\Users\Jakob\AppData\Local\Programs\Microsoft VS Code\resources\app\out\bootstrap-amd.js - c:\Users\Jakob\AppData\Local\Programs\Microsoft VS Code\resources\app\out\bootstrap-fork.js - .

So in my case, the extension doesn't work by default...

jakobjung10, 2023-03-05:
Furthermore, the terminal-output is:

Executing task: npm run webpack 

> vhdl-ls@0.4.0 webpack     
> webpack --mode development

C:\Users\Jakob\Documents\GitHub\rust_hdl_vscode\node_modules\loader-runner\lib\LoaderRunner.js:146
                if(isError) throw e;
                            ^

Error: error:0308010C:digital envelope routines::unsupported
    at new Hash (node:internal/crypto/hash:71:19)
    at Object.createHash (node:crypto:133:10)
    at BulkUpdateDecorator.hashFactory (C:\Users\Jakob\Documents\GitHub\rust_hdl_vscode\node_modules\webpack\lib\util\createHash.js:145:18)
    at BulkUpdateDecorator.update (C:\Users\Jakob\Documents\GitHub\rust_hdl_vscode\node_modules\webpack\lib\util\createHash.js:46:50)
    at SourceMapSource.updateHash (C:\Users\Jakob\Documents\GitHub\rust_hdl_vscode\node_modules\webpack-sources\lib\SourceMapSource.js:233:8)       
    at NormalModule._initBuildHash (C:\Users\Jakob\Documents\GitHub\rust_hdl_vscode\node_modules\webpack\lib\NormalModule.js:870:17)
    at handleParseResult (C:\Users\Jakob\Documents\GitHub\rust_hdl_vscode\node_modules\webpack\lib\NormalModule.js:936:10)
    at C:\Users\Jakob\Documents\GitHub\rust_hdl_vscode\node_modules\webpack\lib\NormalModule.js:1028:4
    at processResult (C:\Users\Jakob\Documents\GitHub\rust_hdl_vscode\node_modules\webpack\lib\NormalModule.js:745:11)
    at C:\Users\Jakob\Documents\GitHub\rust_hdl_vscode\node_modules\webpack\lib\NormalModule.js:809:5 {
  opensslErrorStack: [ 'error:03000086:digital envelope routines::initialization error' ],
  library: 'digital envelope routines',
  reason: 'unsupported',
  code: 'ERR_OSSL_EVP_UNSUPPORTED'
}

Node.js v18.14.1

kraigher, 2023-03-05:
I would suspect the nodejs version. The CI pipeline uses version 16.

jakobjung10, 2023-03-11:
That solved the problem for me. Thank you very much! :)

closed 2023-03-11 by jakobjung10