VHDL-LS/rust_hdl_vscode #67: Error in syntax highlighting when a block does not have anything in its header

LukasVik, 2023-03-10:
This is as expected:

[image]

Removing the content of the header, I would not expect the highlighting to change:

[image]

Test file is attached.  Problem is present in released version as well as in https://github.com/VHDL-LS/rust_hdl_vscode/pull/66 commit `6e85242: Replacing old grammar.`.

bpadalino, 2023-03-10:
I don't think you were using #66.  It turned out that the scoping was incorrect and you would have seen nothing highlighted on the screen with #66.  Same with #68.

Can you please retry and see if you get better results?  Note you will need to replace the [vhdl.tmLanguage](https://github.com/bpadalino/rust_hdl_vscode/blob/vhdl-2019/syntaxes/vhdl.tmLanguage) file.

LukasVik, 2023-03-10:
Yes it seems this is not a problem with #66. And I did indeed build from the wrong commit. Closing this issue.

closed 2023-03-10 by LukasVik