VHDL-LS/rust_hdl_vscode #73: Plugin does not start with VSCode on MacOS

ushi002, 2023-04-26:
It just leaves this error:

[Error - 11:19:38] Starting client failed
Error: spawn Unknown system error -8
	at ChildProcess.spawn (node:internal/child_process:413:11)
	at Object.spawn (node:child_process:720:9)
	at /Users/ludek/.vscode/extensions/hbohlin.vhdl-ls-0.5.0/dist/extension.js:26190:40

jakobjung10, 2023-05-28:
i guess there is no support for MacOS.

In the file "extension.ts" you can see that only binaries for windows and linux are considered.

`const isWindows = process.platform === 'win32';
const languageServerName = isWindows
    ? 'vhdl_ls-x86_64-pc-windows-msvc'
    : 'vhdl_ls-x86_64-unknown-linux-musl';`

kraigher, 2023-05-28:
We do not build binaries for MacOS so that is why there is no support. It should probably be easy to setup if anyone is willing to spend time on it.