VHDL-LS/rust_hdl_vscode #76: [Feature Request] Support custom user-defined types/subtypes

Nabav, 2023-08-23:
LSP v3.17 adds [subtypes](https://microsoft.github.io/language-server-protocol/specifications/lsp/3.17/specification/#typeHierarchy_subtypes). Is it possible to support them in VHDL-LS?

I am experiencing VHDL-LS in VSCode. I see that the custom type is highlighted during declaration.

```vhdl
type my_type_t is (FIRST_STATE, SECOND_STATE);
--   ^^^^^^^^^ <- highlighted
```

But it is _not_ highlighted when defining an instance of that type.

```vhdl
signal my_signal : my_type_t := FIRST_STATE;
--                 ^^^^^^^^^ <- not highlighted
```

The same happens for subtypes. The subtype is highlighted during declaration:

```vhdl
subtype my_int_t is integer range 0 to 63;
--      ^^^^^^^^ <- highlighted
```

But the instance type is _not_ highlighted:

```vhdl
constant MY_CONST : my_int_t := 42;
--                  ^^^^^^^^ <- not highlighted

```

Please let me know, if I am wrong reporting here and the issue should be reported in client side project.

Nabav, 2023-08-29:
Moved to https://github.com/VHDL-LS/rust_hdl/issues/196

Nabav, 2023-11-09:
As mentioned on [sever](https://github.com/VHDL-LS/rust_hdl/issues/196#issuecomment-1711853603) side, the syntax highlighting is not part of the server.

Schottkyc137, 2024-05-26:
Actually, I think that this _does_ belong to the language server 😬.
While it is true that the vscode extension defines the syntax highlighting, the feature that you want doesn't have anything to do with the `subtypes` capability of the LSP. Instead, I believe that what you want are [Semantic Tokens](https://microsoft.github.io/language-server-protocol/specifications/lsp/3.17/specification/#textDocument_semanticTokens). This deals with highlighting types and other language elements which the base implementation herein is not capable of.

Nabav, 2024-05-26:
@kraigher , please consider ☝️, thanks.