VHDL-LS/rust_hdl_vscode #77: Problem with vhdl_ls.toml

Ant1-21, 2023-11-13:
Hello,

I have a problem with linter and vhdl_ls.toml file. Libraries defined in vhdl_ls file are not detected automatically, so the linter display lot of errors.

Each time I open a vhdl file, I need to open vhdl_ls.toml file and save it such that linter finds libraries and removes errors.

I'm using VHDL LS v0.5.0. on debian.

Thank you for your help.

## Vscode
Version : 1.84.2
Validation : 1a5daa3a0231a0fbba4f14db7ec463cf99d7768e
Date : 2023-11-09T10:50:47.800Z
Electron : 25.9.2
ElectronBuildId : 24603566
Chromium : 114.0.5735.289
Node.js : 18.15.0
V8 : 11.4.183.29-electron.0
Système d’exploitation : Linux x64 6.2.0-36-generic snap

kraigher, 2023-11-13:
Could you make a video or provide some reproducer as an isolated project?

kraigher, 2023-11-13:
I have fixed a path inconsistency with the vhdl_ls.toml file in the latest release. You could update and try again.

Ant1-21, 2023-11-14:
Hello,

I tried the new version, but the problem is still present.

I show you my problem in this screenshot and attached project archive: I create a package `my_pkg` where I define a type called `my_type` and which I add in toml file. In my top module, vhdl_ls don't know my package and so `my_type`. I have to open the toml file and save it (even if there is no modification). I have to do it every time I open a file using a custom package.

[image]

[test_vhdl_ls.zip](https://github.com/VHDL-LS/rust_hdl_vscode/files/13346468/test_vhdl_ls.zip)

JHertz5, 2023-11-14:
Hi @Ant1-21, I think I know what your problem is. In your vhdl_ls.toml file, you've set the name of the library as `lib`, but top.vhd, you've declared `use work.my_pkg.all;`. The error message says "No primary unit 'my_pkg' within library 'work'" because you've told the language server that `my_pkg` is in the `lib` library rather than the `work` library, so it can't find `work.my_pkg`. There are two ways to fix this. One is to modify your vhdl_ls.toml file to tell the language server that your files are in the `work` library.
```toml
[libraries]
work.files = [
    "./my_pkg.vhd",
    "./top.vhd",
]
```
The other is to modify top.vhd to call `my_pkg` from the `lib` library.
```vhdl
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

library lib;
use lib.my_pkg.all;

entity top is
    port (
        rst_n    : in  std_logic;
        refclk   : in  std_logic;
        dummy_i  : in  std_logic;
        dummy_o  : out std_logic
    );
end top;

architecture rtl of top is

    component foo is
        port (
            clk   : in  std_logic;
            reset : in  std_logic;
            dummy : out my_type
        );
    end component;

    signal my_bus : my_type;

begin

    foo_inst : component foo
        port map (
            clk   => refclk,
            reset => rst_n,
            dummy => my_bus
        );

end rtl;
```

Ant1-21, 2023-11-14:
Hi,

Ok thank you ! I didn't catch that the name in toml file is important.

Thank you for your help.

kraigher, 2023-11-14:
The problem is that top.vhd is not mapped to any library. work library is actually not a good name for a library as work means "current library" in vhdl. If top.vhd was mapped to lib like the pkg the work.pkg would map to lib.pkg as the current library is lib.

kraigher, 2023-11-14:
Read this to educate yourself on work library: https://insights.sigasi.com/tech/work-not-vhdl-library/

It is surprising how many vhdl developers do not know about it.

closed 2023-11-14 by kraigher

Ant1-21, 2023-11-14:
> Read this to educate yourself on work library: https://insights.sigasi.com/tech/work-not-vhdl-library/
> 
> It is surprising how many vhdl developers do not know about it.

No, I'm not familiar with this language and not a big fan. Moreover, I didn't write the original code, which I've cleaned up and renamed, because I can't share it.

kraigher, 2023-11-14:
The code was fine. The only problem was that the top.vhd file was not mapped to lib in vhdl_ls.toml. if it was mapped to lib then work.pkg would be ok as pkg would be in the same library as top.vhd