VHDL-LS/rust_hdl_vscode #80: Libraries removed from `vhdl_ls.toml` still appearing in problems window

pintert3, 2024-01-10:
It seems the VHDL LS is viewing bugs and errors in files that aren't even in my `vhdl_ls.toml`. This is so quite annoying cause I'm having to go to those files to fix errors that are moreover third party libraries in some cases. this is how it looks:
[image]

None of those files are in the active `vhdl_ls.toml`. Previously I had UNISIM amongst them, and it also had an error for me, despite setting it as a 3rd party. The common thing with these files though, is that they were all once in a `vhdl_ls.toml` file in another project. So I don't know why they are still sort of cached in the LS extension. really need help.

JHertz5, 2024-01-10:
Does this problem persist if you restart VHDL LS, or reload the VS Code window? 
[image]

pintert3, 2024-01-10:
Yes it does. It's been doing it for at least the last 2-3 months

kraigher, 2024-01-10:
There is no caching in VHDL-LS at all. Every time you start it loads the files. Just must be using the wrong toml file, check the log to see which toml files are loaded.

pintert3, 2024-01-10:
How can I check those logs? Also, could it be related to why the `is_third_party` option isn't actually stopping lints for third party libraries? I thought it would help to silence them. It's like a lot of things are messed up somewhere.

pintert3, 2024-01-10:
So after moving things around, it turns out the first problem was that when there are errors in any of the files in the TOML file, it seemed to keep errors from other files. I have no idea why though. But putting together the `vunit` library well (adding missing *deprecated files and so on), the errors reduced to strictly those in the files in the work library. I'm at least glad it worked, though I don't think one error should pull errors from allover.

pintert3, 2024-01-10:
So it turns out that after reverting the change I had made to UNISIM.VPKG to correct the unused variable error, it returns in my list of `PROBLEMS` window. So it's not all solved.

pintert3, 2024-03-01:
Sorry for this. I finally figured out the perpetrator. I had an old `.vhdl_ls.toml` stored in my home directory. Hadn't noticed it for some reason. I was under the impression that the `vhdl_ls.toml` in the local directory was the first priority and would override any other existing. Documentation on [VHDL-LS/rust-hdl](https://github.com/VHDL-LS/rust_hdl) says otherwise, and was the reason I had those old libraries.
Thanks for the help, guys.