VHDL-LS/rust_hdl_vscode #81: Indentation rules are case sensitive

Sv3n, 2024-02-09:
VHDL is not case sensitive, but the indentation rules are. I made a change to `language-configuration.json` that fixes this.

Old:

```json
    "indentationRules": {
        "increaseIndentPattern": "\\b(begin|is|port|else|loop|generate|then)\\b",
        "decreaseIndentPattern": "^\\s*(elsif|else|end|begin)\\b"
    }
```

Updated:

```json
    "indentationRules": {
        "increaseIndentPattern": {
            "pattern": "\\b(begin|is|port|else|loop|generate|then)\\b",
            "flags": "i"
        },
        "decreaseIndentPattern": {
            "pattern": "^\\s*(elsif|else|end|begin)\\b",
            "flags": "i"
        }
    }
```

I learned about the flags option here in case you are wondering: https://github.com/microsoft/vscode/issues/74493.
Perhaps this is something that would be useful for others as well?

Side note: the only detail that does not really work to my liking is `WHEN` statements, but that is more of a personal preference, I was trying to get them to format like this, but I don't see how that can be done with the available indentationRules handholds in vscode.

```vhdl
CASE selector IS
    WHEN "00" =>
        out <= option0;
    WHEN "01" =>
        out <= option1;
    WHEN "10" =>
        out <= option2;
    WHEN OTHERS =>
        out <= option3;
END CASE;
```

JHertz5, 2024-02-09:
I could be mistaken, but I don't think this has anything to do with VHDL_LS. Language configuration is a built-in feature of VS Code. This repository is for a third-party VHDL language server. If you want a tool that can do VHDL formatting, I suggest that you look at this https://vhdl-style-guide.readthedocs.io/en/3.21.0/overview.html

Sv3n, 2024-02-10:
For a moment, I thought I had my tabs crossed, but it is actually here, so I think I've posted this in the right place (note the rust_hdl_vscode vs rust_hdl):

https://github.com/VHDL-LS/rust_hdl_vscode/blob/master/language-configuration.json

JHertz5, 2024-02-10:
I stand corrected, apologies!

Schottkyc137, 2024-03-03:
This looks like a nice addition. Could you open a Pull Request with your change?

Sv3n, 2024-03-10:
Here's the pull request: https://github.com/VHDL-LS/rust_hdl_vscode/pull/83

Schottkyc137, 2024-03-10:
Closed with #83

closed 2024-03-10 by Schottkyc137