VHDL-LS/rust_hdl_vscode #101: the extension does'nt working at all

em1983, 2024-10-28:
Hi, 
I used to work with this extension for a while, but after updating to **v0.7.0**, I have been having a lot of issues like the ones below. I tried going back to an earlier version to continue, but that version also displayed errors. Occasionally, after a few attempts with various versions, if it starts to work, the next day, after reopening the Vscode, everything starts over and....

[Error - 4:04:36 PM] 
**The VHDL LS server crashed 5 times in the last 3 minutes. The server will not be restarted. See the output for more information.**
[Error - 4:04:36 PM] 
**VHDL LS client: couldn't create connection to server.**
  Message: Pending response rejected since connection got disposed
  Code: -32097 
[Error - 4:04:36 PM] 
**Restarting server failed**
  Message: Pending response rejected since connection got disposed
  Code: -32097

Schottkyc137, 2024-10-28:
Hi,
Could you provide more information about your operating system and / or code you are working with? Unfortunately, the error you describe could have a lot of reasons

em1983, 2024-10-29:
Hej Hej
I appreciate your response. I am using a **64-bit / Windows 11 Enterprise** system, and every time I close VScode and reopen it, ´
i should try different VHDL_LS versions until it works. 
Could it be because it is my working laptop that I had to switch to Windows 11 recently?

em1983, 2025-02-06:
Hi again,
This issue persists. I should spend a couple of hours **switching between different versions** until it starts working because it is not fixed on my end. While this approach does not always work, there are times when it does.

**Err:**
[Error - 2:28:31 PM] The VHDL LS server crashed 5 times in the last 3 minutes. The server will not be restarted. See the output for more information.
[Error - 2:28:31 PM] VHDL LS client: couldn't create connection to server.
  Message: Pending response rejected since connection got disposed
  Code: -32097 
[Error - 2:28:31 PM] Restarting server failed
  Message: Pending response rejected since connection got disposed
  Code: -32097 
[Error - 2:28:31 PM] Server process exited with code 101.

Schottkyc137, 2025-02-06:
How are you installing `vhdl_ls`? Are you using the version bundled through the VSCode extension or did you download the version yourself?

em1983, 2025-02-07:
Hi, tnx for your response.
Yes, I'm using the VScode extension to install it. I never tried to download it directly. Could that possibly be a solution?

Schottkyc137, 2025-02-07:
Could you do the following, please?
In VSCode, on the project that you use the language server on, go to "Output" (commonly on the bottom of the window, in the same tab group as i.e., Terminal) and choose "VHDL-LS Client" on the pop-down menu on the right. Could you paste the output that you see here?

em1983, 2025-02-07:
**Here you are:**

Checking for language server executable in c:\Users\EM4489\.vscode\extensions\hbohlin.vhdl-ls-0.7.0\server\vhdl_ls
Found version 0.80.0
Using embedded language server
Checking for updates...
Language server started
Current vhdl_ls version: 0.80.0
Latest vhdl_ls version: 0.83.1
Fetching https://github.com/VHDL-LS/rust_hdl/releases/download/v0.83.1/vhdl_ls-x86_64-pc-windows-msvc.zip
Writing c:\Users\EM4489\.vscode\extensions\hbohlin.vhdl-ls-0.7.0\server\install\0.83.1\vhdl_ls-x86_64-pc-windows-msvc.zip
Server download complete
Extracting c:\Users\EM4489\.vscode\extensions\hbohlin.vhdl-ls-0.7.0\server\install\0.83.1\vhdl_ls-x86_64-pc-windows-msvc.zip to c:\Users\EM4489\.vscode\extensions\hbohlin.vhdl-ls-0.7.0\server\vhdl_ls\0.83.1
Server extracted to c:\Users\EM4489\.vscode\extensions\hbohlin.vhdl-ls-0.7.0\server\vhdl_ls\0.83.1
Language server update finished.

Schottkyc137, 2025-02-07:
If you close VSCode and re-open it, does the message stay the same?

em1983, 2025-02-07:
NO, got the error now
When you close and reopen it, start sending failure notifications continuously. the **VHDL-LS Clinet OUTPUT** message is like this: 

Checking for language server executable in c:\Users\EM4489\.vscode\extensions\hbohlin.vhdl-ls-0.7.0\server\vhdl_ls
Found version 0.83.1
Using embedded language server

Schottkyc137, 2025-02-07:
And is the error output that you posted everything that you see (I.e., the whole log) Usually, there should be more information concerning the error

em1983, 2025-02-07:
When I select **VHDL LS** I can see all of those crash and failure notifications, but if I select **VHDL-LS Client**, that is the whole log I have.

From here, I should continue to **switch between diff versions** until one of them can be enabled.

em1983, 2025-02-07:
But why, when it started to work, the version was like: **Found version 0.80.0** and after close and reopen changes to: **Found version 0.83.1** 
Is this trying to tell us something?

Schottkyc137, 2025-02-07:
> When I select **VHDL LS** I can see all of those crash and failure notifications, but if I select **VHDL-LS Client**, that is the whole log I have.
> 
> From here, I should continue to **switch between diff versions** until one of them can be enabled.

Can you show me the whole log? I believe that what you have shown is not the entire log, but I might be wrong.

> But why, when it started to work, the version was like: **Found version 0.80.0** and after close and reopen changes to: **Found version 0.83.1** Is this trying to tell us something?

There are two different versions at play:
- The version of the VSCode extension (0.7.0) changes very rarely.
- The version of the language server (aka the backend of the VSCode extension) changes much faster. Therefore, there is an automated way to update the language server (which you have experienced now).

The issue that you are experiencing likely has something to do with the language server version, not the VSCode version

em1983, 2025-02-10:
Hej 
Sorry for the late response. I don't have access to my work computer during the weekend.

If it works (after many tries and errors), it looks like a short message, but once it's not working, the message in the VHDL-LS client will be very short, but the **VHDL-LS tab** will have a long message of a panicked message with a lot of not-clear CHAR that I will attach below:

thread '<unnamed>' panicked at vhdl_lang\src\data\contents.rs:359:38:
byte index 98 is not a char boundary; it is inside '¡' (bytes 97..99) of `.¹a¯«)í&.~àímJØÃ¾)X	ÄÕõBV:°×?ËPÆX»Çº<Á\çmÁ!þÅGþ63h%¡_m¾+ÕÂÉnÀÁ;Hllpì_bÊqÝd"Hjm&ÛðöóÌyXlxVHYEB     400     170k¸0Þýp¬±¥¬ØºÂÆTe8Ê(Æ¹kp%2dáåa6w`[...]
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
Loaded Installation configuration file: c:\Users\EM4489\.vscode\extensions\hbohlin.vhdl-ls-0.7.0\server\vhdl_ls\0.83.1\vhdl_ls-x86_64-pc-windows-msvc\bin\../vhdl_libraries\vhdl_ls.toml
Loaded workspace root configuration file: c:\Users\EM4489\GIT\fpga_dev\vhdl_ls.toml
[Error - 9:57:12 AM] Server process exited with code 101.
[Error - 9:57:12 AM] Server initialization failed.
  Message: Pending response rejected since connection got disposed
  Code: -32097 
[Info  - 9:57:12 AM] Connection to server got closed. Server will restart.
true
[Error - 9:57:12 AM] VHDL LS client: couldn't create connection to server.
  Message: Pending response rejected since connection got disposed
  Code: -32097 
Loaded Installation configuration file: c:\Users\EM4489\.vscode\extensions\hbohlin.vhdl-ls-0.7.0\server\vhdl_ls\0.83.1\vhdl_ls-x86_64-pc-windows-msvc\bin\../vhdl_libraries\vhdl_ls.toml
Loaded workspace root configuration file: c:\Users\EM4489\GIT\fpga_dev\vhdl_ls.toml
thread '<unnamed>' panicked at vhdl_lang\src\data\contents.rs:359:38:
byte index 98 is not a char boundary; it is inside '¡' (bytes 97..99) of `.¹a¯«)í&.~àímJØÃ¾)X	ÄÕõBV:°×?ËPÆX»Çº<Á\çmÁ!þÅGþ63h%¡_m¾+ÕÂÉnÀÁ;Hllpì_bÊqÝd"Hjm&ÛðöóÌyXlxVHYEB     400     170k¸0Þýp¬±¥¬ØºÂÆTe8Ê(Æ¹kp%2dáåa6w`[...]
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
[Error - 9:57:19 AM] Server process exited with code 101.
[Error - 9:57:19 AM] Server initialization failed.
  Message: Pending response rejected since connection got disposed
  Code: -32097 
[Info  - 9:57:19 AM] Connection to server got closed. Server will restart.
true
[Error - 9:57:19 AM] VHDL LS client: couldn't create connection to server.
  Message: Pending response rejected since connection got disposed
  Code: -32097 
thread '<unnamed>' panicked at vhdl_lang\src\data\contents.rs:359:38:
byte index 98 is not a char boundary; it is inside '¡' (bytes 97..99) of `.¹a¯«)í&.~àímJØÃ¾)X	ÄÕõBV:°×?ËPÆX»Çº<Á\çmÁ!þÅGþ63h%¡_m¾+ÕÂÉnÀÁ;Hllpì_bÊqÝd"Hjm&ÛðöóÌyXlxVHYEB     400     170k¸0Þýp¬±¥¬ØºÂÆTe8Ê(Æ¹kp%2dáåa6w`[...]
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
Loaded Installation configuration file: c:\Users\EM4489\.vscode\extensions\hbohlin.vhdl-ls-0.7.0\server\vhdl_ls\0.83.1\vhdl_ls-x86_64-pc-windows-msvc\bin\../vhdl_libraries\vhdl_ls.toml
Loaded workspace root configuration file: c:\Users\EM4489\GIT\fpga_dev\vhdl_ls.toml
[Error - 9:57:22 AM] Server initialization failed.
  Message: Pending response rejected since connection got disposed
  Code: -32097 
[Info  - 9:57:22 AM] Connection to server got closed. Server will restart.
true
[Error - 9:57:22 AM] VHDL LS client: couldn't create connection to server.
  Message: Pending response rejected since connection got disposed
  Code: -32097 
[Error - 9:57:22 AM] Restarting server failed
  Message: Pending response rejected since connection got disposed
  Code: -32097 
[Error - 9:57:23 AM] Server process exited with code 101.
thread '<unnamed>' panicked at vhdl_lang\src\data\contents.rs:359:38:
byte index 98 is not a char boundary; it is inside '¡' (bytes 97..99) of `.¹a¯«)í&.~àímJØÃ¾)X	ÄÕõBV:°×?ËPÆX»Çº<Á\çmÁ!þÅGþ63h%¡_m¾+ÕÂÉnÀÁ;Hllpì_bÊqÝd"Hjm&ÛðöóÌyXlxVHYEB     400     170k¸0Þýp¬±¥¬ØºÂÆTe8Ê(Æ¹kp%2dáåa6w`[...]
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
[Error - 9:57:26 AM] Server process exited with code 101.
Loaded Installation configuration file: c:\Users\EM4489\.vscode\extensions\hbohlin.vhdl-ls-0.7.0\server\vhdl_ls\0.83.1\vhdl_ls-x86_64-pc-windows-msvc\bin\../vhdl_libraries\vhdl_ls.toml
[Error - 9:57:26 AM] Server initialization failed.
  Message: Pending response rejected since connection got disposed
  Code: -32097 
[Info  - 9:57:26 AM] Connection to server got closed. Server will restart.
true
[Error - 9:57:26 AM] VHDL LS client: couldn't create connection to server.
  Message: Pending response rejected since connection got disposed
  Code: -32097 
[Error - 9:57:26 AM] Restarting server failed
  Message: Pending response rejected since connection got disposed
  Code: -32097 
thread '<unnamed>' panicked at vhdl_lang\src\data\contents.rs:359:38:
byte index 98 is not a char boundary; it is inside '¡' (bytes 97..99) of `.¹a¯«)í&.~àímJØÃ¾)X	ÄÕõBV:°×?ËPÆX»Çº<Á\çmÁ!þÅGþ63h%¡_m¾+ÕÂÉnÀÁ;Hllpì_bÊqÝd"Hjm&ÛðöóÌyXlxVHYEB     400     170k¸0Þýp¬±¥¬ØºÂÆTe8Ê(Æ¹kp%2dáåa6w`[...]
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
[... cut ...]
  Code: -32097 
[Error - 9:57:29 AM] The VHDL LS server crashed 5 times in the last 3 minutes. The server will not be restarted. See the output for more information.
[Error - 9:57:29 AM] VHDL LS client: couldn't create connection to server.
  Message: Pending response rejected since connection got disposed
  Code: -32097 
[Error - 9:57:29 AM] Restarting server failed
  Message: Pending response rejected since connection got disposed
  Code: -32097 
[Error - 9:57:29 AM] Server process exited with code 101.

em1983, 2025-02-10:
I think I found it.

To map all our subfolders in a folder and also to prevent duplicate a folder name, we use a **double star char (**)** to cover all of our subfolders like this, and it causes the error. 

**[libraries]**
**hard_fifo.files = ['C:\Users\EM4489\GIT\fpga_dev\projects\hard_fifo\**\*.vhd']**

As soon as I comment it out and reload the window any time, it works perfectly. Is this approach correct? or there is a safer method to use instead of those (**)?

Schottkyc137, 2025-02-10:
The usage of asterisks shouldn't be a problem. I suspect that the real problem is instead in one of the VHDL files in `hard_fifo`.

em1983, 2025-02-10:
I have tried with all subfolders, like:

[libraries]
hard_fifo.files = ['C:\Users\EM4489\GIT\fpga_dev\projects\hard_fifo \source\ *.vhd']
or 
[libraries]
hard_fifo.files = ['C:\Users\EM4489\GIT\fpga_dev\projects\hard_fifo \simulation\ *.vhd']
or 
[libraries]
hard_fifo.files = ['C:\Users\EM4489\GIT\fpga_dev\projects\hard_fifo \IPs\ *.vhd']

It works immediately and shows unmaped and unused signals, etc. Supposing I have 3 main subfolders, like **src,sim,pkg**. How should we map them without replicating the main folder name?