VHDL-LS/rust_hdl_vscode #99: VSCode Extension TerosHDL Error

woeichee, 2024-10-28:
I am facing an errorbelow after installing TerosHDL. Any idea to fix it?

thread '<unnamed>' panicked at vhdl_lang\src\named_entity\formal_region.rs:101:17:
internal error: entered unreachable code

Schottkyc137, 2024-10-28:
Hi,

Could you provide an example that causes your code to panic? This error could come from a lot of different places although I suspect that this is due to the same reason as https://github.com/VHDL-LS/rust_hdl/issues/348?

woeichee, 2024-10-28:
> Hi,
> 
> Could you provide an example that causes your code to panic? This error could come from a lot of different places although I suspect that this is due to the same reason as [VHDL-LS/rust_hdl#348](https://github.com/VHDL-LS/rust_hdl/issues/348)?

I don't have any idea which part of my code.  I just open the my workspace that contain a lot of vhdl files.  Then this message keeps happening.  So, i need to close all first and then open 1 by 1?  After that, how to deal with it?

Schottkyc137, 2024-10-28:
Unfortunately, there is not a lot you can do at the moment as this is an error on the language server side. Again, I suspect that this has something to do with entity instantiations that are not signals or constants (e.g., types or functions), but I can't be sure of this without more context. If you want to help me, you could try and compile a minimum reproducible example from your projects that generates the panic.