VHDL-LS/rust_hdl_vscode #103: Problems with embedded executable auto-update after v0.83.1 was published

ALL-SPACE-Jukka, 2024-12-16:
I'm having issues using the VHDL LS extension on VS Code after version 0.83.1 of rust_hdl was published.

When I open VS Code, I get no VS code functionality and an error saying
```
[Error - 10:56:25] VHDL LS client: couldn't create connection to server.
Launching server using command /mnt/nfs/home/jhertzog/.vscode-server/extensions/hbohlin.vhdl-ls-0.7.0/server/vhdl_ls/0.83.1/vhdl_ls-x86_64-unknown-linux-musl/bin/vhdl_ls failed. Error: spawn /mnt/nfs/home/jhertzog/.vscode-server/extensions/hbohlin.vhdl-ls-0.7.0/server/vhdl_ls/0.83.1/vhdl_ls-x86_64-unknown-linux-musl/bin/vhdl_ls ENOENT
```
while the output for VHDL-LS Client says
```
Checking for language server executable in /mnt/nfs/home/jhertzog/.vscode-server/extensions/hbohlin.vhdl-ls-0.7.0/server/vhdl_ls
Found version 0.83.1
Using embedded language server
```
`ENOENT` usually means a directory or file not existing, so I checked the path listed above and the path `0.83.1/vhdl_ls-x86_64-unknown-linux-musl/bin/` exists, but it is empty.

To force it to re-download the executable, I deleted the `0.83.1` directory and restarted VS code. This time, I get the following output
```
Checking for language server executable in /mnt/nfs/home/jhertzog/.vscode-server/extensions/hbohlin.vhdl-ls-0.7.0/server/vhdl_ls
No language server installed
Current vhdl_ls version: 0.0.0
Latest vhdl_ls version: 0.83.1
Fetching https://github.com/VHDL-LS/rust_hdl/releases/download/v0.83.1/vhdl_ls-x86_64-unknown-linux-musl.zip
Writing /mnt/nfs/home/jhertzog/.vscode-server/extensions/hbohlin.vhdl-ls-0.7.0/server/install/0.83.1/vhdl_ls-x86_64-unknown-linux-musl.zip
Server download complete
Extracting /mnt/nfs/home/jhertzog/.vscode-server/extensions/hbohlin.vhdl-ls-0.7.0/server/install/0.83.1/vhdl_ls-x86_64-unknown-linux-musl.zip to /mnt/nfs/home/jhertzog/.vscode-server/extensions/hbohlin.vhdl-ls-0.7.0/server/vhdl_ls/0.83.1
Server extracted to /mnt/nfs/home/jhertzog/.vscode-server/extensions/hbohlin.vhdl-ls-0.7.0/server/vhdl_ls/0.83.1
Using embedded language server
Checking for updates...
Language server started
Current vhdl_ls version: 0.64.0
Latest vhdl_ls version: 0.83.1
Fetching https://github.com/VHDL-LS/rust_hdl/releases/download/v0.83.1/vhdl_ls-x86_64-unknown-linux-musl.zip
Writing /mnt/nfs/home/jhertzog/.vscode-server/extensions/hbohlin.vhdl-ls-0.7.0/server/install/0.83.1/vhdl_ls-x86_64-unknown-linux-musl.zip
Server download complete
Extracting /mnt/nfs/home/jhertzog/.vscode-server/extensions/hbohlin.vhdl-ls-0.7.0/server/install/0.83.1/vhdl_ls-x86_64-unknown-linux-musl.zip to /mnt/nfs/home/jhertzog/.vscode-server/extensions/hbohlin.vhdl-ls-0.7.0/server/vhdl_ls/0.83.1
Error when extracting server
Error: ETXTBSY: text file is busy, open '/mnt/nfs/home/jhertzog/.vscode-server/extensions/hbohlin.vhdl-ls-0.7.0/server/vhdl_ls/0.83.1/vhdl_ls-x86_64-unknown-linux-musl/bin/vhdl_ls'
Cannot remove /mnt/nfs/home/jhertzog/.vscode-server/extensions/hbohlin.vhdl-ls-0.7.0/server/vhdl_ls/0.83.1: Error: ENOTEMPTY: directory not empty, rmdir '/mnt/nfs/home/jhertzog/.vscode-server/extensions/hbohlin.vhdl-ls-0.7.0/server/vhdl_ls/0.83.1/vhdl_ls-x86_64-unknown-linux-musl/bin'
Error: ETXTBSY: text file is busy, open '/mnt/nfs/home/jhertzog/.vscode-server/extensions/hbohlin.vhdl-ls-0.7.0/server/vhdl_ls/0.83.1/vhdl_ls-x86_64-unknown-linux-musl/bin/vhdl_ls'
Language server update finished.
```
It looks to me like the VHDL-LS extension is trying to simultaneously download and write two copies of the executable, causing conflicts with itself.
However this time, I do get some VS code functionality (but I don't think it's the latest version since there are a lot of errors flagged on valid code that weren't there in the past) and the following output for VHDL-LS Client
```
Loaded Installation configuration file: /mnt/nfs/home/jhertzog/.vscode-server/extensions/hbohlin.vhdl-ls-0.7.0/server/vhdl_ls/0.83.1/vhdl_ls-x86_64-unknown-linux-musl/bin/../vhdl_libraries/vhdl_ls.toml
Loaded workspace root configuration file: /mnt/nfs/data/users/jhertzog/s2000/main/vhdl_ls.toml
```

However, when I restart VS Code it's back to `VHDL LS client: couldn't create connection to server.`

Any ideas what might be going on? Any help would be greatly appreciated.

JHertz5, 2024-12-16:
Thanks very much!

ALL-SPACE-Jukka, 2024-12-16:
> I'll try to investigate this week; preferably today

That would be amazing, thanks very much for the quick response. This tool is so useful that I've become very dependent on it 😅.

Schottkyc137, 2024-12-16:
For the meantime, you could also manually download a release [here](https://github.com/VHDL-LS/rust_hdl/releases/tag/v0.83.0)

ALL-SPACE-Jukka, 2024-12-17:
Hi @Schottkyc137.

Thanks for that, I was confused about why 0.64.0 was showing up in the log!

Unfortunately, the problem is still occurring. I've done some digging and I think I know why.

Judging from the output logs in VS Code, it looks like the extension expects to find the embedded executable at the following path `/mnt/nfs/home/jhertzog/.vscode-server/extensions/hbohlin.vhdl-ls-0.7.0/server/vhdl_ls/0.83.1/vhdl_ls-x86_64-unknown-linux-musl/bin/vhdl_ls`, but there is nothing at this path. The executable has been installed in a path without the `vhdl_ls-x86_64-unknown-linux-musl` directory:
```
$ ls /mnt/nfs/home/jhertzog/.vscode-server/extensions/hbohlin.vhdl-ls-0.7.0/server/vhdl_ls/0.83.1/vhdl_ls-x86_64-unknown-linux-musl/bin/vhdl_ls
ls: cannot access '/mnt/nfs/home/jhertzog/.vscode-server/extensions/hbohlin.vhdl-ls-0.7.0/server/vhdl_ls/0.83.1/vhdl_ls-x86_64-unknown-linux-musl/bin/vhdl_ls': No such file or directory
$ ls /mnt/nfs/home/jhertzog/.vscode-server/extensions/hbohlin.vhdl-ls-0.7.0/server/vhdl_ls/0.83.1/bin/vhdl_ls
/mnt/nfs/home/jhertzog/.vscode-server/extensions/hbohlin.vhdl-ls-0.7.0/server/vhdl_ls/0.83.1/bin/vhdl_ls
```

If I set the extension to use a user-defined path, I can use the executable stored at the second location no problem.

I've downloaded the zip files from the rust_hdl server to check their contents, and the `vhdl_ls-x86_64-unknown-linux-musl` directory is omitted from the latest release's zip file.

```
$ unzip -q vhdl_ls-x86_64-unknown-linux-musl-0-83-1.zip -d ./vhdlls_0_83_1/
$ tree -L 1 ./vhdlls_0_83_1
./vhdlls_0_83_1
├── bin
└── vhdl_libraries
$ unzip -q vhdl_ls-x86_64-unknown-linux-musl-0-83-0.zip -d ./vhdlls_0_83_0/
$ tree -L 2 ./vhdlls_0_83_0
./vhdlls_0_83_0
└── vhdl_ls-x86_64-unknown-linux-musl
    ├── bin
    └── vhdl_libraries
```

I think that this is the source of the problem. The extension unzips the latest release, but because it's directory structure is different from what is expected, the executable doesn't end up at the right path. Hopefully this is easily fixed?

ALL-SPACE-Jukka, 2024-12-19:
It looks like this issue is covered by an issue in the executable's repository https://github.com/VHDL-LS/rust_hdl/issues/357. The issue has been resolved for me after deleting the `0.83.1` directory to force the extension to redownload the executable. Thanks very much for your help @Schottkyc137.

closed 2024-12-19 by ALL-SPACE-Jukka